MorneSaunders360/Solar-Sunsynk #5: Missing sensors

Nurgus, 2023-06-29:
I was using the sensors _loadOrEpsPower and _gridormeterpower for house load and grid load respectively. Having updated today, all the sensors are renamed and I can't find these two values. 

None of the numbers for any entities seem to match.

[image]

Nurgus, 2023-06-29:
I'm guessing that _instantaneous_grid_i_o and _instantaneous_load are  the sensors I need but they're a wildly different.

MorneSaunders360, 2023-06-29:
Hi Nurgus,

I wanted to address a few points concerning the current Sunsynk API's integration with Home Assistant. I have identified several issues that I believe may be leading to the instability you've been experiencing. 

Firstly, I found that the initial setup of the device as a service was incorrect, which may be contributing to the hanging/crashing problem with Home Assistant. 

Secondly, it appears that some extraneous sensors were inadvertently included in the device display, which don't pertain to the Sunsynk integration. This is an area that certainly needs rectifying to avoid potential confusion and system overload.

Additionally, I realized that the integration needs to accommodate multiple solar systems, as a plant could potentially contain more than one inverter. 

Furthermore, I am currently enhancing the Sunsynk integration to function effectively with the Energy Dashboard. I understand that this feature was not working as expected before, which may have caused some inconvenience.

I deeply appreciate your patience as I work on resolving these issues. Please feel free to share any additional feedback you might have. Your insights are invaluable in helping me improve this integration. I will keep you updated as progress is made and aim to rectify the situation as promptly as possible.

MorneSaunders360, 2023-06-29:
We also get the data now straight on inverter level
 
instantaneous_load = loadOrEpsPower
_instantaneous_grid_i_o = gridOrMeterPower

Let me know if I'm wrong, I'm still trying to figure out where to map each sensor correctly

MorneSaunders360, 2023-06-29:
Update

Instantaneous Grid I/O Total = gridOrMeterPower
Instantaneous Load = loadOrEpsPower

Nurgus, 2023-06-30:
Hi, I'm impressed with how this integration is coming along and excited to get started using the service to control my Sunsynk gear!

Instantaneous Grid I/O Total = gridOrMeterPower
Instantaneous Load = loadOrEpsPower

The problem I have is that these sensors are giving me totally different readings to the ones I get from the sunsynk app.

In the screenshot, my house is pulling 7923w yet the Instantaneous Load sensor says 292w

From the grid it's pulling 26w and the Instantaneous Grid I/O Total sensor shows -1945w, These sensors are always wrong.

Instantaneous Battery and Instantaneous Generation sensors are always correct.

Let me know if I can provide any more information.

MorneSaunders360, 2023-06-30:
So I think what might have happened was the refresh timer was 30 to 60 seconds behind and might be the reason for why the data is not matching your app and in home assistant.
I will take a look at the Instantaneous Grid I/O Total as it might read the wrong sensor will keep you updated
This is how my data looks like:
[image]

Nurgus, 2023-06-30:
I've been refreshing the app and comparing the numbers to this integration almost continuously since yesterday and there's been no moment where those two sensors have been anywhere near to matching up. Meanwhile the generation and battery sensors are spot on.

Something's definitely not right but I don't know how to diagnose it further.

Nurgus, 2023-06-30:
[image]
[image]

This illustrates the problem. In Sunsynk connect the load is around 450w and then jumps to around 1500w with not much in between. In Home Assistant the load history is very spiky and doesn't match up at all.

Nurgus, 2023-06-30:
> Integrations > Solar Sunsynk > Go into the Integration and Click on re-download the integration.

Done, no improvement.

> other then that, I can give you the old branch with the old integration code (This will be the last resort if all fails)?

Nah I want to help crack the problem. Let me know if you think of anything to test or any information I can provide.

MorneSaunders360, 2023-06-30:
Hmmm.... okay I'm going to use sunsynk virtual user and then I will check their system and see if I can see anything out of the ordinary.

MorneSaunders360, 2023-06-30:
I can see the differences now
[image]

MorneSaunders360, 2023-06-30:
Okay, updated the integration, check if your data is now updated to the correct values

Nurgus, 2023-06-30:
Oh yes wow, that's fixed it. All sensors looking good.  What was the problem?

MorneSaunders360, 2023-06-30:
So it seems like there is more than one place you can get those sensor values. The ones I selected were the individual values for load and grid and the correct ones are calculated

closed 2023-06-30 by Nurgus