MorneSaunders360/Solar-Sunsynk #11: Can't switch Grid Charge 1-6 or Gen Charge 1-6 on

Nurgus, 2023-10-18:
Is there a guide for all the options the service can control? What does mondayOn do for example? There's nothing in the official UI that seems to correspond.

I'm using this in a script. It changes the other settings but the time1on true/false options (including 1 to 6) don't take effect. The existing settings remain.

cap1 to cap6 do take effect.

```
sequence:
  - service: solar_sunsynk.set_solar_settings
    data:
      sn: "xxxxx"
      sellTime1: "01:00"
      sellTime2: "05:00"
      sellTime3: "12:30"
      sellTime4: "13:00"
      sellTime5: "17:00"
      sellTime6: "21:00"
      cap1: "20"
      cap2: "20"
      cap3: "20"
      cap4: "20"
      cap5: "20"
      cap6: "20"
      time1on: false
      time2on: true
      time3on: true
      time4on: false
      time5on: false
      time6on: false
      sysWorkMode: "2"
```

MorneSaunders360, 2023-10-20:
Hey Nurgus,

Sorry for the delay in response; I was away on my honeymoon. I've noticed there's no official guide for each of the sensors. I'll work on setting up a guide and updating the readme. Additionally, I'll investigate why the other settings aren't updating.

Thanks for your patience!

MorneSaunders360, 2023-10-25:
Hello Nurgus,

I've uploaded a Service Guide to our repository, which you can find here:
[Service Guide on GitHub](https://github.com/MorneSaunders360/Solar-Sunsynk/blob/main/ServiceGuide.md).

Please review the guide, especially the mappings between the service and the front end. I hope this will enhance your understanding of how the Sunsynk service operates.

Nurgus, 2023-11-03:
Many thanks for the excellent service guide. Regarding the issue with time1on..time6on not taking effect - are you able to reproduce it or is it just me?

MorneSaunders360, 2023-11-04:
I did see that these time1on..time6on settings is not taking any effect on the website even looking at the raw payload when updating the settings

MorneSaunders360, 2023-11-04:
In order to set time1on..time6on, you need to add genTime1on..genTime6on to the service.
It seems like they are working in a pair in order to be updated

[image]
[image]

closed 2023-12-16 by MorneSaunders360