MorneSaunders360/Solar-Sunsynk #13: TypeError: 'NoneType' object is not iterable

acormack, 2024-03-29:
Hi Morne

I have been running Solar-Sunsynk for many months without problem - Many Thanks!  But it now has started hanging with the error below.  If I restart Home Assistant it works for a few minutes, then the error recurrs.

Please can you help?  

I am running v1.05 on 3.6Kw inverter in UK.

=======================================
This error originated from a custom integration.
Logger: custom_components.solar_sunsynk
Source: helpers/update_coordinator.py:318
integration: Solar Sunsynk ([documentation](https://github.com/MorneSaunders360/Solar-Sunsynk), [issues](https://github.com/MorneSaunders360/Solar-Sunsynk/issues))
First occurred: 10:12:45 (2 occurrences)
Last logged: 10:13:15

Unexpected error fetching solar_sunsynk data: 'NoneType' object is not iterable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 318, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solar_sunsynk/coordinator.py", line 29, in _async_update_data
    for invertor in jsondata:
TypeError: 'NoneType' object is not iterable

alz41, 2024-03-30:
This aswell

Logger: custom_components.solar_sunsynk.sunsynkapi
Source: custom_components/solar_sunsynk/sunsynkapi.py:61
integration: Solar Sunsynk ([documentation](https://github.com/MorneSaunders360/Solar-Sunsynk), [issues](https://github.com/MorneSaunders360/Solar-Sunsynk/issues))
First occurred: 01:44:28 (4 occurrences)
Last logged: 02:04:31

Error Connecting: HTTPSConnectionPool(host='api.sunsynk.net', port=443): Max retries exceeded with url: /api/v1/plants?page=1&limit=10&name=&status= (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1000)')))

MorneSaunders360, 2024-04-03:
Hello,

Firstly, I'd like to extend my apologies for the delay in addressing your concerns. My schedule has been notably dense due to commitments to a new project, which has temporarily diverted my attention from monitoring the issue log.

Regarding the issue with the Solar Sunsynk integration, I am aware that Sunsynk has been updating their backend, which may be influencing the current situation. Rest assured, I am prioritizing a review of this matter to understand the specifics of the problem.

I will keep you updated with any developments and will expedite an update as soon as possible to resolve this issue. Thank you for your patience and understanding.

Best regards,

MorneSaunders360, 2024-04-03:
Hi,

I have published a new release, please update and let me know if you still get any errors.

If you do get any errors, I have added additional loggers to help us detect where the error lies.

acormack, 2024-04-04:
> Hi,
> 
> I have published a new release, please update and let me know if you still get any errors.
> 
> If you do get any errors, I have added additional loggers to help us detect where the error lies.

Hi Morne,

Thnks for the new version.  I did the update and still got errors.  This is the log output:

2024-04-04 08:56:14.067 ERROR (MainThread) [custom_components.solar_sunsynk.sunsynkapi] Calling get_plant_data with args: ()
2024-04-04 08:56:14.067 ERROR (MainThread) [custom_components.solar_sunsynk.sunsynkapi] Error while getting plant data: 'NoneType' object is not subscriptable
2024-04-04 08:56:14.067 ERROR (MainThread) [custom_components.solar_sunsynk] Unexpected error fetching solar_sunsynk data: 'NoneType' object is not iterable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solar_sunsynk/coordinator.py", line 29, in _async_update_data
for invertor in jsondata:
TypeError: 'NoneType' object is not iterable
2024-04-04 08:56:14.143 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 0.626 seconds (success: False)
2024-04-04 08:56:20.015 ERROR (MainThread) [custom_components.solar_sunsynk.sunsynkapi] Calling get_plant_data with args: ()
2024-04-04 08:56:20.015 ERROR (MainThread) [custom_components.solar_sunsynk.sunsynkapi] Error while getting plant data: 'NoneType' object is not subscriptable
2024-04-04 08:56:20.016 ERROR (MainThread) [custom_components.solar_sunsynk] Unexpected error fetching solar_sunsynk data: 'NoneType' object is not iterable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solar_sunsynk/coordinator.py", line 29, in _async_update_data
for invertor in jsondata:
TypeError: 'NoneType' object is not iterable
2024-04-04 08:56:20.021 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 0.627 seconds (success: False)
2024-04-04 08:56:30.822 ERROR (MainThread) [custom_components.solar_sunsynk.sunsynkapi] Calling get_plant_data with args: ()
2024-04-04 08:56:30.823 ERROR (MainThread) [custom_components.solar_sunsynk.sunsynkapi] Error while getting plant data: 'NoneType' object is not subscriptable
2024-04-04 08:56:30.823 ERROR (MainThread) [custom_components.solar_sunsynk] Unexpected error fetching solar_sunsynk data: 'NoneType' object is not iterable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solar_sunsynk/coordinator.py", line 29, in _async_update_data
for invertor in jsondata:
TypeError: 'NoneType' object is not iterable
2024-04-04 08:56:30.837 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 0.598 seconds (success: False)

MorneSaunders360, 2024-04-04:
Hmmm this is quite a strange one, can you maybe uninstall the sunsynk integration and reinstall it and also when reinstalling please check the region and the interval, I know sunsynk has been moving people's solar systems over to a new region and the endpoints does have an impact on the synsynk integration. Looking forward to your next response

acormack, 2024-04-04:
Hi Morne

Thanks for the quick response. I uninstalled the Device/Service and removed the HACS download.  I checked the directory was deleted, then reinstalled.

I entered my username/password and Region 2 (I don't know how I check if this is correct???)

I started the service and it initially got readings from the server four times (30 seconds apart) , but then it got these errors from the server every 30 seconds after that.  Could it be the default 30 second interval is too short?

Alec
==============================
Log output below.....

2024-04-04 16:28:51.893 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 2.531 seconds (success: True)
2024-04-04 16:29:23.995 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 2.633 seconds (success: True)
2024-04-04 16:29:55.576 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 2.214 seconds (success: True)
2024-04-04 16:30:27.619 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 2.256 seconds (success: True)
2024-04-04 16:31:00.019 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 2.656 seconds (success: True)
2024-04-04 16:31:29.564 ERROR (SyncWorker_18) [custom_components.solar_sunsynk.sunsynkapi] Error Connecting: HTTPSConnectionPool(host='api.sunsynk.net', port=443): Max retries exceeded with url: /api/v1/plants?page=1&limit=10 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1000)')))
2024-04-04 16:31:29.570 ERROR (MainThread) [custom_components.solar_sunsynk] Unexpected error fetching solar_sunsynk data: 'NoneType' object is not iterable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solar_sunsynk/coordinator.py", line 29, in _async_update_data
for invertor in jsondata:
TypeError: 'NoneType' object is not iterable
2024-04-04 16:31:29.579 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 0.217 seconds (success: False)
2024-04-04 16:31:59.460 ERROR (SyncWorker_4) [custom_components.solar_sunsynk.sunsynkapi] Error Connecting: HTTPSConnectionPool(host='api.sunsynk.net', port=443): Max retries exceeded with url: /api/v1/plants?page=1&limit=10 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1000)')))
2024-04-04 16:31:59.465 ERROR (MainThread) [custom_components.solar_sunsynk] Unexpected error fetching solar_sunsynk data: 'NoneType' object is not iterable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solar_sunsynk/coordinator.py", line 29, in _async_update_data
for invertor in jsondata:
TypeError: 'NoneType' object is not iterable
2024-04-04 16:31:59.472 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 0.110 seconds (success: False)
2024-04-04 16:32:29.485 ERROR (SyncWorker_0) [custom_components.solar_sunsynk.sunsynkapi] Error Connecting: HTTPSConnectionPool(host='api.sunsynk.net', port=443): Max retries exceeded with url: /api/v1/plants?page=1&limit=10 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1000)')))
2024-04-04 16:32:29.493 ERROR (MainThread) [custom_components.solar_sunsynk] Unexpected error fetching solar_sunsynk data: 'NoneType' object is not iterable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solar_sunsynk/coordinator.py", line 29, in _async_update_data
for invertor in jsondata:
TypeError: 'NoneType' object is not iterable
2024-04-04 16:32:29.496 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 0.133 seconds (success: False)
2024-04-04 16:32:59.508 ERROR (SyncWorker_20) [custom_components.solar_sunsynk.sunsynkapi] Error Connecting: HTTPSConnectionPool(host='api.sunsynk.net', port=443): Max retries exceeded with url: /api/v1/plants?page=1&limit=10 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1000)')))
2024-04-04 16:32:59.517 ERROR (MainThread) [custom_components.solar_sunsynk] Unexpected error fetching solar_sunsynk data: 'NoneType' object is not iterable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solar_sunsynk/coordinator.py", line 29, in _async_update_data
for invertor in jsondata:
TypeError: 'NoneType' object is not iterable
2024-04-04 16:32:59.530 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 0.167 seconds (success: False)

Nurgus, 2024-04-04:
Regarding region, Sunsynk moved all acounts to region 1 recently so the only correct answer is region 2.

MorneSaunders360, 2024-04-05:
Thanks, Nurgus for assisting.
I have removed the regions selector and now looking at the region 2 (https://api.sunsynk.net/)
Also changed the default interval to 60 seconds instead of 30 seconds.
I have also managed to replicate the bug and have made some changes and testing the changes in a test environment.
Will keep you updated on the test results.

MorneSaunders360, 2024-04-05:
Okay, I have created a beta version of the sunsynk integration for you to test as well.
You can redownload the integration and select the beta version (v1.0.7-beta)
[image]
I'm also running this version and waiting for it to give me any errors.

acormack, 2024-04-05:
Hi Morne

I have the beta version running now and it looks promising. 4 successful transactions in a row so far.  I will continue to test

Alec

2024-04-05 14:25:19.956 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 4.587 seconds (success: True)
2024-04-05 14:30:22.839 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 2.417 seconds (success: True)
2024-04-05 14:35:24.857 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 2.435 seconds (success: True)
2024-04-05 14:40:26.661 DEBUG (MainThread) [custom_components.solar_sunsynk] Finished fetching solar_sunsynk data in 2.240 seconds (success: True)

acormack, 2024-04-06:
Hi Morne

I have been running the beta version every 30 seconds for the last 18 hours. 2187 successful API calls have been made with no errors. I think you have solved the problem! 

Many Thanks

Alec

MorneSaunders360, 2024-04-06:
Hi

Perfect, I also have been running the integration with 10 solar systems and I did not receive any errors.
I will do a full release

MorneSaunders360, 2024-04-06:
I have published a new version to the main branch.
Latest version is v1.0.8

closed 2024-04-07 by MorneSaunders360