MorneSaunders360/Solar-Sunsynk #15: 1.0.8 fails to start

freaksdotcom, 2024-04-09:
HA reports that 1.0.8 fails to setup.

Reverting to 1.0.6 fixes the problem.

From the logs:
```
2024-04-09 09:03:32.160 ERROR (MainThread) [custom_components.solar_sunsynk.sunsynkapi] Calling get_plant_data with args: ()
2024-04-09 09:03:32.160 ERROR (MainThread) [custom_components.solar_sunsynk.sunsynkapi] Error while getting plant data: 'NoneType' object is not subscriptable
2024-04-09 09:03:32.160 ERROR (MainThread) [custom_components.solar_sunsynk] Unexpected error fetching solar_sunsynk data: 'NoneType' object is not iterable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solar_sunsynk/coordinator.py", line 29, in _async_update_data
for invertor in jsondata:
TypeError: 'NoneType' object is not iterable
```

MorneSaunders360, 2024-04-09:
Hi

Apologies for the integration not starting up.
Can you provide me with a bit more detail?
Example can you uninstall the integration and reinstall version 1.0.6 and while you are busy with the setup what region you were using? (I know Sunsynk has move all of the solar systems for region 2)
If you are using version 1.0.6 uninstall it and download v1.0.8 and run the setup and if you get any errors please provide those logs as well.

Verion v1.0.8 does have a few breaking changes where we have removed the regions selector and only using region 2 since the cross over to the new region

freaksdotcom, 2024-04-10:
I'm using the integration with a Sol-Ark 15K, not a Sunsynk. It was set up using region 1.

MorneSaunders360, 2024-04-11:
Ahh okay, that makes more sense now, I wrote this integration for Sunsynk exclusively. 
I can however add an extra check box that will allow for other solar systems that is using region 1.

MorneSaunders360, 2024-04-11:
Okay, I added two regions call Sunsynk and PowerView, you can safely update to v1.0.10

freaksdotcom, 2024-04-16:
Thank you! It appears to be working.

closed 2024-04-16 by freaksdotcom