MorneSaunders360/Solar-Sunsynk #16: Region 2 error

bacardiiii, 2024-04-28:
Thank you for your great integration. when i check my logs i see i keep getting a error

Logger: custom_components.solar_sunsynk.sunsynkapi
Source: custom_components/solar_sunsynk/sunsynkapi.py:36
integration: Solar Sunsynk ([documentation](https://github.com/MorneSaunders360/Solar-Sunsynk), [issues](https://github.com/MorneSaunders360/Solar-Sunsynk/issues))
First occurred: 17:19:27 (244 occurrences)
Last logged: 17:47:13

Region 2

Seeing as Sunsynk had been changing things i deleted the add on and reinstalled, however the error still exists. Any ideas how i can rectify this.

Thanks

MorneSaunders360, 2024-04-29:
Hi,

I've looked into the error you were experiencing and discovered that I accidentally left a debug log in my code. I apologize for the oversight and have now removed the log entry.

I will release a new update asap.

bacardiiii, 2024-04-30:
No problem. I've updated and its now clear. Thanks for your help

closed 2024-04-30 by bacardiiii