MorneSaunders360/Solar-Sunsynk #17: error adding repository

SHadley1138, 2024-07-09:
every time I try to add the repository, I get this error "Repository MorneSaunders360/Solar-Sunsynk not found". very odd

MorneSaunders360, 2024-07-09:
Hi hi

That is really odd. Please ensure you have the latest hacs and Home assistant installed.

We have also added a button in our repo to add the repo to home assistant.

[image]

It might also be that there is a bug in hacs or home assistant when adding the repo, does this also happen when you add another repo to hacs?

SHadley1138, 2024-07-09:
apologies for the url video post

MorneSaunders360, 2024-07-09:
Cool, thanks for the video, I remove my repo from HACS and used the button on this repo and I'm also able to replicate the issue. 
I will investigate and report back on a fix asap.

MorneSaunders360, 2024-07-09:
Okay, I have found the issue with the button and made a patch for it, you can try the add button on this repo again.
Let me know if this patch works for you?

SHadley1138, 2024-07-09:
Sadly not….  Using the button, it connects to my HA, then I get this error
Repository MorneSaunders360/Solar-Sunsynk not found

MorneSaunders360, 2024-07-09:
Hmm.. Okay, let's try to do it manually.
Go to hacs and head into the integrations section.

The click on the hamburger menu icon(the three dots) and select custom repository.
[image].

Then add the repo using this url
https://github.com/MorneSaunders360/Solar-Sunsynk

Select the integration option and then click on the add button.
[image]

Then it should appear on the list.
[image]

[image]

SHadley1138, 2024-07-10:
Hi,
Ive given it a go manually, but I get this error...

￼

> On 9 Jul 2024, at 22:45, Morne ***@***.***> wrote:
> 
> https://github.com/MorneSaunders360/Solar-Sunsynk

MorneSaunders360, 2024-07-11:
This is a weird one. Hmm, do you know what version of hacs you are currently using?

SHadley1138, 2024-07-11:
It reckons 1.34.0

MorneSaunders360, 2024-07-12:
Hmm, okay, so my version is the same. Maybe the home assistant is outdated?

SHadley1138, 2024-07-12:
No, I’m running 14.7.2

MorneSaunders360, 2024-07-12:
Okay lets then check the HA Core version this is my current version
[image]