MorneSaunders360/Solar-Sunsynk #21: Integration won’t load

bacardiiii, 2025-01-07:
Hi

I still have the issue I reported when I was on beta, I have switched off beta  and running 2025.1.1. Is this something you’re also having an issue with now or is it just me?

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solar_sunsynk/__init__.py", line 34, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2278, in async_forward_entry_setups
    await integration.async_get_platforms(platforms)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1185, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    platforms.update(self._load_platforms(platform_names))
                     ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1098, in _load_platforms
    platform_name: self._load_platform(platform_name)
                   ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1272, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1304, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solar_sunsynk/sensor.py", line 9, in <module>
    from homeassistant.const import ENERGY_KILO_WATT_HOUR, PERCENTAGE, POWER_WATT
ImportError: cannot import name 'ENERGY_KILO_WATT_HOUR' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)

MorneSaunders360, 2025-01-07:
Hi

I haven't updated my home assistant yet, I will update mine tomorrow and let you know if I experience the same problem.

laicheungkee, 2025-01-08:
Seems I have the same problem.

I installed all updates available from Home Assistant Core to any other integrations/add-ons available, which I do regret now.

Initially the battery level could not be shown.

[image]

When I was trying to reconfigure the integration, it encountered problem.

[image]

The log I could gather is as the following:

Logger: homeassistant.config_entries
Source: config_entries.py:640
First occurred: 11:26:44 AM (2 occurrences)
Last logged: 1:49:05 PM

Error setting up entry lai.alexander@gmail.com for solar_sunsynk
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solar_sunsynk/__init__.py", line 34, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2278, in async_forward_entry_setups
    await integration.async_get_platforms(platforms)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1185, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    platforms.update(self._load_platforms(platform_names))
                     ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1098, in _load_platforms
    platform_name: self._load_platform(platform_name)
                   ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1272, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1304, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solar_sunsynk/sensor.py", line 9, in <module>
    from homeassistant.const import ENERGY_KILO_WATT_HOUR, PERCENTAGE, POWER_WATT
ImportError: cannot import name 'ENERGY_KILO_WATT_HOUR' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)

I am sorry that I am not a very technical people, if more information is needed, please do let me know and I will try my best to provide.  Thanks.

Alexander Lai

MorneSaunders360, 2025-01-08:
Thanks, I also upgrade my home assistant and re-installed the integration and mine is also breaking, will work on a patch, and keep you guys updated

MorneSaunders360, 2025-01-08:
I patch the Sunsynk integration v1.0.12.
ENERGY_KILO_WATT_HOUR and POWER_WATT was deprecated in Home Assistant 2025.1

laicheungkee, 2025-01-08:
Thanks @MorneSaunders360 .  What should I do now? Wait for the update becomes available?

MorneSaunders360, 2025-01-08:
You go into HACS and just click on Sunsynk integration and then click on update information and a popup should appear for an update
[image]

laicheungkee, 2025-01-08:
Thank you @MorneSaunders360 , the problem is fixed! Wish you will have a very good day!

[image]

bacardiiii, 2025-01-08:
Thanks for the brilliant work. All working fine

closed 2025-01-08 by bacardiiii