MorneSaunders360/Solar-Sunsynk #22: Unverified HTTPS request is being made to host 'api.sunsynk.net

bacardiiii, 2025-01-09:
Hi 

I checked the logs this morning and have this appearing, is this ok to leave? I can't remember it being there before

This error originated from a custom integration.

Logger: py.warnings
Source: custom_components/solar_sunsynk/sunsynkapi.py:56
integration: Solar Sunsynk (documentation, issues)
First occurred: 8 January 2025 at 19:01:34 (7983 occurrences)
Last logged: 10:17:13

/usr/local/lib/python3.13/site-packages/urllib3/connectionpool.py:1064: InsecureRequestWarning: Unverified HTTPS request is being made to host 'api.sunsynk.net'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/1.26.x/advanced-usage.html#ssl-warnings warnings.warn(

MorneSaunders360, 2025-01-09:
Hi,

Thanks, I made a patch for this warning, will push the update now

bacardiiii, 2025-01-09:
Brilliant thanks again for a great integration

closed 2025-01-09 by bacardiiii