MorneSaunders360/Solar-Sunsynk #23: Solarsynk returns null readings alternate with correct readings every few minutes.

Selseymech, 2025-01-20:
Solarsynk returns null readings alternate with correct readings every few minutes.I have reinstalled the add -on renewed the long access and double checked the Sunsynk log in criteria.The add- on was working well up to just a few days ago.

MorneSaunders360, 2025-01-20:
Hi,

I also seem to get these null readings, will check tomorrow and let you know.

Selseymech, 2025-01-20:
Hi morne      Many thanks for responding - great integration and your help is very much appreciated-thank youGeoff MellettSent from my iPad

MorneSaunders360, 2025-01-21:
Hey 

Just a quick update, I'm making some code changes and is currently in the testing phase, as this bug happens after a while.

Selseymech, 2025-01-21:
> Just a quick update, I'm making some code changes and I'm currently in the

Thank you for the update and your patience Morne it is very much
appreciated !

Selseymech, 2025-01-22:
> I made some changes to the integration which will now update the sensor every 5 minutes, instead of every 1 minute resulting in the integration failing .
>
> New version v1.0.14

Good afternoon Morne,
                            The changes you made seem to be solid -no null returns all day, well done and thank you. I am a bit confused though- the integration you have successfully amended is Solar-sunsynk but the integration I was having trouble with was "Solarsynk" which is working well again too??  I'm probably missing something here but can't see how one would cure the other!!!

Best regards and thank you again.

Geoff Mellett

MorneSaunders360, 2025-01-22:
Good afternoon Geoff,

Thank you for your kind words—I’m glad to hear the changes are yielding positive results. Regarding the confusion, you’re absolutely right to ask. While the Solar-Sunsynk and Solarsynk integrations are distinct, it’s possible that some shared functionality or underlying dependencies between the two were inadvertently impacted or improved during the amendment process.

Best regards,
Morne

closed 2025-01-22 by MorneSaunders360