MorneSaunders360/Solar-Sunsynk #24: Setting grid charge settings

tfitzjohn, 2025-02-09:
Hi,

These system settings are fantastic for automating battery dumping or charging -cthanks!

I'm trying to use the set system settings to change the grid charge settings Onan event trigger from another automation. But the settings aren't being applied.

I'm using the following as described in GitHub

time5on: true
sellTime5: "18:00"
cap5: "100"

MorneSaunders360, 2025-02-09:
Hi,  

I encountered the same issue when trying to set only one of the `sellTime` values. It turns out that you need to set all of them at once; otherwise, the changes won’t take effect.  

Here’s an example of how it should be configured:  

```json
{
    "time1on": "false",
    "time2on": "false",
    "time3on": "false",
    "time4on": "false",
    "time5on": "true",
    "time6on": "false",
    "sellTime1": "01:00",
    "sellTime2": "05:00",
    "sellTime3": "12:30",
    "sellTime4": "13:00",
    "sellTime5": "8:03",
    "sellTime6": "21:00",
    "cap1": "70",
    "cap2": "70",
    "cap3": "70",
    "cap4": "70",
    "cap5": "100",
    "cap6": "70"
}
```
Let me know if you need any help if the above does not work for you.

tfitzjohn, 2025-02-09:
Hi, the times and caps are updating but not the tick boxes

Here's the yaml I'm using (hiding the serial number)

action: solar_sunsynk.set_solar_settings
metadata: {}
data:
  sn: "**********"
  time1on: true
  time2on: true
  time3on: true
  time4on: true
  time5on: true
  time6on: true
  sellTime1: "00:00"
  sellTime2: "05:30"
  sellTime3: "12:30"
  sellTime4: "13:00"
  sellTime5: "17:10"
  sellTime6: "23:30"
  cap1: "100"
  cap2: "100"
  cap3: "100"
  cap4: "100"
  cap5: "100"
  cap6: "100"

MorneSaunders360, 2025-02-09:
Try this and let me know if it works

action: solar_sunsynk.set_solar_settings
metadata: {}
data:
  sn: "**********"
  time1on: "true"
  time2on: "true"
  time3on: "true"
  time4on: "true"
  time5on: "true"
  time6on: "true"
  sellTime1: "00:00"
  sellTime2: "05:30"
  sellTime3: "12:30"
  sellTime4: "13:00"
  sellTime5: "17:10"
  sellTime6: "23:30"
  cap1: "100"
  cap2: "100"
  cap3: "100"
  cap4: "100"
  cap5: "100"
  cap6: "100"

tfitzjohn, 2025-02-09:
Nope and nope 😔

MorneSaunders360, 2025-02-11:
Have you tried the complete example and payload, I'm wondering if there nees to be an extra value that needs to be set to enable grid charge?

{
    "sn": "**********",
    "safetyType": "2",
    "battMode": "-1",
    "solarSell": "0",
    "pvMaxLimit": "8000",
    "energyMode": "0",
    "peakAndVallery": "1",
    "sysWorkMode": "2",
    "sellTime1": "01:00",
    "sellTime2": "05:00",
    "sellTime3": "12:30",
    "sellTime4": "13:00",
    "sellTime5": "17:00",
    "sellTime6": "21:00",
    "sellTime1Pac": "8000",
    "sellTime2Pac": "8000",
    "sellTime3Pac": "8000",
    "sellTime4Pac": "8000",
    "sellTime5Pac": "8000",
    "sellTime6Pac": "8000",
    "cap1": "70",
    "cap2": "70",
    "cap3": "70",
    "cap4": "70",
    "cap5": "70",
    "cap6": "70",
    "sellTime1Volt": "49",
    "sellTime2Volt": "49",
    "sellTime3Volt": "49",
    "sellTime4Volt": "49",
    "sellTime5Volt": "49",
    "sellTime6Volt": "49",
    "zeroExportPower": "0",
    "solarMaxSellPower": "9000",
    "mondayOn": true,
    "tuesdayOn": true,
    "wednesdayOn": true,
    "thursdayOn": true,
    "fridayOn": true,
    "saturdayOn": true,
    "sundayOn": true,
    "time1on": "false",
    "time2on": "false",
    "time3on": "false",
    "time4on": "false",
    "time5on": "false",
    "time6on": "false",
    "genTime1on": "false",
    "genTime2on": "false",
    "genTime3on": "false",
    "genTime4on": "false",
    "genTime5on": "false",
    "genTime6on": "false"
}

tfitzjohn, 2025-02-12:
> ID: <MorneSaunders360/Solar-Sunsynk/issues/24/2651711183 @ github. com>

Yes, this works!

I'll have a look tonight at what stops it working when removed ...

Tom Fitzjohn
07851214523

Sent via OnMail

--00000000000000cca8d4cc0f1c5efa
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=utf-8

        <div>          Yes, this works!</div><div><br></div><div>I'll have =
a look tonight at what stops it working when removed ...<br>          <br> =
               <div class=3D"webmail_signature">        <div><span style=3D=
"color:rgb(153, 153, 153)">Tom Fitzjohn</span></div><div><span style=3D"col=
or:rgb(153, 153, 153)">07851214523</span></div>        <br><div>  <span sty=
le=3D"color: #999999;" data-mce-style=3D"color: #999999;" data-color=3D"#99=
9999">Sent via</span> <a target=3D"_blank" rel=3D"noopener noreferrer" href=
=3D"https://www.onmail.com?referrer=3Dsignature" data-mce-href=3D"https://w=
ww.onmail.com?referrer=3Dsignature">OnMail</a></div>      </div>           =
   <br>    <br>    <div class=3D"webmail_quote">      <div class=3D"webmail=
_attr">        On Tue, Feb 11, 2025 at 6:49 PM Morne ***@***.***=
ub.com&gt; wrote:      </div>      <blockquote class=3D"webmail_quote" styl=
e=3D"          margin: 0px 0px 0px 0.8ex;          border-left: 1px solid r=
gb(204, 204, 204);          padding-left: 1ex;        ">        <p></p><p d=
ir=3D"auto">Have you tried the complete example and payload, I'm wondering =
if there nees to be an extra value that needs to be set to enable grid char=
ge?</p><p dir=3D"auto">{<br>"sn": "**********",<br>"safetyType": "2",<br>"b=
attMode": "-1",<br>"solarSell": "0",<br>"pvMaxLimit": "8000",<br>"energyMod=
e": "0",<br>"peakAndVallery": "1",<br>"sysWorkMode": "2",<br>"sellTime1": "=
01:00",<br>"sellTime2": "05:00",<br>"sellTime3": "12:30",<br>"sellTime4": "=
13:00",<br>"sellTime5": "17:00",<br>"sellTime6": "21:00",<br>"sellTime1Pac"=
: "8000",<br>"sellTime2Pac": "8000",<br>"sellTime3Pac": "8000",<br>"sellTim=
e4Pac": "8000",<br>"sellTime5Pac": "8000",<br>"sellTime6Pac": "8000",<br>"c=
ap1": "70",<br>"cap2": "70",<br>"cap3": "70",<br>"cap4": "70",<br>"cap5": "=
70",<br>"cap6": "70",<br>"sellTime1Volt": "49",<br>"sellTime2Volt": "49",<b=
r>"sellTime3Volt": "49",<br>"sellTime4Volt": "49",<br>"sellTime5Volt": "49"=
,<br>"sellTime6Volt": "49",<br>"zeroExportPower": "0",<br>"solarMaxSellPowe=
r": "9000",<br>"mondayOn": true,<br>"tuesdayOn": true,<br>"wednesdayOn": tr=
ue,<br>"thursdayOn": true,<br>"fridayOn": true,<br>"saturdayOn": true,<br>"=
sundayOn": true,<br>"time1on": "false",<br>"time2on": "false",<br>"time3on"=
: "false",<br>"time4on": "false",<br>"time5on": "false",<br>"time6on": "fal=
se",<br>"genTime1on": "false",<br>"genTime2on": "false",<br>"genTime3on": "=
false",<br>"genTime4on": "false",<br>"genTime5on": "false",<br>"genTime6on"=
: "false"<br>}</p><p style=3D"font-size:small;-webkit-text-size-adjust:none=
;color:#666;">=E2=80=94<br>
/github.com/MorneSaunders360/Solar-Sunsynk/issues/24#issuecomment-265171118=
3">view it on GitHub</a>, or <a href=3D"https://github.com/notifications/un=
subscribe-auth/BGHCTFHXFO4L5CKNQV52UQD2PJA3DAVCNFSM6AAAAABWYQWODOVHI2DSMVQW=
IX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDMNJRG4YTCMJYGM">unsubscribe</a>.<br>You are=
 receiving this because you modified the open/close state.<span style=3D"co=
lor: transparent; font-size: 0; display: none; visibility: hidden; overflow=
: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height: 0; mso=
2651711183</span><span>@</span><span>github</span><span>.</span><span>com&g=
t;</span></span></p><script ***@***.***": "htt=
ction","target": "https://github.com/MorneSaunders360/Solar-Sunsynk/issues/=
24#issuecomment-2651711183","url": "https://github.com/MorneSaunders360/Sol=
ar-Sunsynk/issues/24#issuecomment-2651711183","name": "View Issue"},"descri=
ption": "View this Issue on GitHub","publisher": ***@***.***": "Organization","=
name": "GitHub","url": "https://github.com"}}]</script>      </blockquote> =
   </div>        </div>
--00000000000000cca8d4cc0f1c5efa--

closed 2025-02-23 by MorneSaunders360