MorneSaunders360/Solar-Sunsynk #25: Home assistant - failed setup

tfitzjohn, 2025-02-11:
Hi,

I've been using solar sunsynk for months - it's fantastic. But this morning I encountered this error (attached) in home assistant, any ideas please?

[image]

Thanks!

MorneSaunders360, 2025-02-11:
Hi, what version of home assistant are you using? and what version of the integration are you also using?

MorneSaunders360, 2025-02-11:
I'm running the same versions, have you tried to reinstall the SunSynk Intergration and log in from setup? 

[image]

tfitzjohn, 2025-02-11:
Solved!

My sincere apologies - I had turned MFA on on my sunsynk account yesterday and I didn't put 2 and 2 together!

I have turned it off and the error has gone away.

Thanks!

closed 2025-02-12 by MorneSaunders360