MorneSaunders360/Solar-Sunsynk #32: All Data unavailable - list index out of range

alexmonaghan, 2025-08-11:
Have restarted HA, currently on HA 2025.7.4, all data points are showing as unavailable, HA logs show 

Logger: custom_components.solar_sunsynk
Source: custom_components/solar_sunsynk/coordinator.py:144
integration: Solar Sunsynk ([documentation](https://github.com/MorneSaunders360/Solar-Sunsynk), [issues](https://github.com/MorneSaunders360/Solar-Sunsynk/issues))
First occurred: 08:47:41 (1 occurrence)
Last logged: 08:47:41
IndexError for inverter sunsynk_IDString_IDString while processing statistics: list index out of range. (followed by long JSON string with personal details, can provide removing personal data if required.)

Happy to run further diags if required. Data is being provided on sunsynk.net and also on the app, so logins and Sunsynk's DC are good, has been working well for a while. Can currently access Sunsynk API via my own PHP application.

MorneSaunders360, 2025-08-11:
Hi,

Thank you for bringing this to our attention. I’ve checked my logs and can confirm we’re able to replicate the issue. We’ll work on a fix and provide an update as soon as possible.

MorneSaunders360, 2025-08-11:
I’ve shipped an update. Please pull the latest version and let me know if you run into any other issues.

alexmonaghan, 2025-08-11:
It's loaded OK and is giving data into HA, so it looks like we're good, if you hear nothing in the next few days, please feel free to make as complete. I have a small LCD on my desk displaying Sunsync (and other helpful data) so I'll soon see if there's an issue.

Thanks for the prompt response.